Ysurac/openmptcprouter #4046: VPS1 to VPS2 Wireguard VPN for OMR1 to OMR2 LAN Static Routes

Network-Traditions, 2025-08-19:
## Expected Behavior
Establish a wireguard or other vpn to route LAN traffic between two OpenMPTCProuter systems that leverages the aggregated OMR tunnels.  Ideally, the intra-OMR vpn would establish a bidirectional static route link between VPS1 and VPS2, which would forward private RFC 1918 packets to OMR1 and OMR respectively.

## Current Behavior
The expected behavior has been achieved between one OpenMPTCProuter system and one pfSense system using a wireguard vpn as documented in #2123 using Glorytun TCP as OMR's default VPN with a LAN interface static route for the pfSense LAN targets using OMR's 10.255.255.2 interface as the gateway.  Subsequently OMR LAN packets received at the pfSense wireguard interface has a source IP of 10.255.255.2.  This requires an allowed IP entry on the pfSense wireguard peer configuration of 10.255.255.0/24 as @Ysurac, @highfreq and @highvolt-dev established:

> Conclusion (and actual fix):
10.255.255.0/24 must be added to AllowedIPs section of wireguard client configuration in addition to your OMR lan IP (192.168.12.0/24) for me.

The challenge in deploying the same approach for two OMR systems (i.e. OMR1 <-> VPS1 <-> wg1VPN <-> VPS2 <->OMR2), both OMRs and VPSs have the same Glorytun TCP IP addresses of 10.255.255.2 and 10.255.255.1 respectively.  Using static routing between the systems requires all IP addresses to be unique.  In this scenario, the Glorytun TCP (tun0) interface of OMR1 and OMR2 both have the IP address 10.255.255.2/32 and the Glorytun TCP (gt-tun0) interface of VPS1 and VPS2 both have the IP address 10.255.255.1/30.

It appears that the OMR settings wizard controls the Glorytun TCP IP addresses for OMR and the VPS.  All that is required to make this work is an ability to uniquely configure the Glorytun TCP IP addresses throughout all statically routed connections (i.e. OMR1 tun0=10.255.255.2; VPS1 gt-tun0=10.255.255.1;  OMR2 tun0=10.255.255.4; VPS1 gt-tun0=10.255.255.3).  Where and/or how are the IP addresses for the Glorytun TCP interfaces set?  If they can be modified will they survive OMR's settings wizard execution?  Once the Glorytun TCP interface IP addresses are set on OMR and its respective VPS within the 10.255.255.0/24 subnet, do they every change (i.e. start as 10.255.255.2, but end up as 10.255.255.3 as a part of normal operation causing a duplicate address conflict if 10.255.255.3 was assigned to another system)?

## Specifications

  - OpenMPTCProuter version: 0.63-snapshot-6.12
  - OpenMPTCProuter VPS version: 0.1032 6.12.15-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: https://cloudfanatic.net/
  - OpenMPTCProuter platform: GL-MT-6000 (mediatek-filogic-glinet_gl-mt6000-squashfs-sysupgrade.bin) August 13, 2025

Ysurac, 2025-08-19:
Using one VPS for 2 or more OMR routers already allow to enable client to client.
Yes, using 2 VPS will make some conflicts, difficult to know who you really want to join on VPS1 or VPS2...

Network-Traditions, 2025-08-20:
It's been a while since I've reviewed OMR's progress in detail.  I recall the concept of connecting multiple OMR's to a single VPS being raised, but at that time, such a configuration wasn't supported.  I'll spend some time researching multiple OMR single VPS configurations and would appreciate any guidance regarding information I should review.  I'm using OMR for its intended purpose of bringing datacenter internet connectivity to facilities that for one reason or another cannot duplicate datacenter internet service quality and bandwidth.  The multiple OMR connectivity is to interconnect the respective facilities of a multi-location operation (i.e. trusted intranet connections).  Thanks for the update!

closed 2025-11-24 by github-actions[bot]