Ysurac/openmptcprouter #4049: Build failed with omr_feed_src develop on ustream-ssl.

tiborkiss, 2025-08-20:
## Expected Behavior
`
OMR_TARGET="bpi-r4" OMR_FEED_SRC="develop" OMR_KERNEL="6.6" ./build.sh -j1 V=s 2>&1 | tee build.log
`

## Current Behavior
The build failed in ustream-ssl
`
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
mbedcrypto_library  .... 
mbedtls_library   ....
mbedx509_library   ...
`

## Possible Solution

## Steps to Reproduce the Problem

  1. clone the openmptcprouter repo (main) and try to build
`
OMR_TARGET="bpi-r4" OMR_FEED_SRC="develop" OMR_KERNEL="6.6" ./build.sh -j1 V=s 2>&1 | tee build.log
`
  2. After so many things built, here is the relevant build log
`
make[3]: Entering directory '/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/package/libs/ustream-ssl'
mkdir -p /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/dl
SHELL= flock /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/tmp/.ustream-ssl-2024.07.28~99bd3d2b.tar.zst.flock -c '    /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/scripts/download.pl "/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/dl" "ustream-ssl-2024.07.28~99bd3d2b.tar.zst" "d3a07151ec361cc4483406a6d01ebbf6ddf627e54f5a5b6676f265614797ee84" "" || (     echo "Checking out files from the git repository..."; mkdir -p /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/tmp/dl && cd /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/tmp/dl && rm -rf ustream-ssl-2024.07.28~99bd3d2b && [ \! -d ustream-ssl-2024.07.28~99bd3d2b ] && git clone  https://git.openwrt.org/project/ustream-ssl.git ustream-ssl-2024.07.28~99bd3d2b && (cd ustream-ssl-2024.07.28~99bd3d2b && git checkout 99bd3d2b167ccdffb6de072d02c380cb37b23e33) && export TAR_TIMESTAMP=`cd ustream-ssl-2024.07.28~99bd3d2b && git log -1 --format='\''@%ct'\''` && echo "Generating formal git archive (apply .gitattributes rules)" && (cd ustream-ssl-2024.07.28~99bd3d2b && git config core.abbrev 8 && git archive --format=tar HEAD --output=../ustream-ssl-2024.07.28~99bd3d2b.tar.git) && tar --ignore-failed-read -C ustream-ssl-2024.07.28~99bd3d2b -f ustream-ssl-2024.07.28~99bd3d2b.tar.git -r .git .gitmodules 2>/dev/null && rm -rf ustream-ssl-2024.07.28~99bd3d2b && mkdir ustream-ssl-2024.07.28~99bd3d2b && tar -C ustream-ssl-2024.07.28~99bd3d2b -xf ustream-ssl-2024.07.28~99bd3d2b.tar.git && (cd ustream-ssl-2024.07.28~99bd3d2b && git submodule update --init --recursive --  && rm -rf .git .gitmodules) && echo "Packing checkout..." &&          tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c ustream-ssl-2024.07.28~99bd3d2b |  zstd -T0 --ultra -20 -c > /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/tmp/dl/ustream-ssl-2024.07.28~99bd3d2b.tar.zst && mv /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/tmp/dl/ustream-ssl-2024.07.28~99bd3d2b.tar.zst /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/dl/ && rm -rf ustream-ssl-2024.07.28~99bd3d2b;  )    '
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/ustream-ssl-2024.07.28~99bd3d2b.tar.zst
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0^M100 14632  100 14632    0     0  49340      0 --:--:-- --:--:-- --:--:-- 49432
touch /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/ustream-ssl-mbedtls/ustream-ssl-2024.07.28~99bd3d2b/.prepared_a666c3eebf6946f85b105b4493acfc72_18f1e190c5d53547fed41a3eaa76e9e9_check
. /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/include/shell.sh; zstdcat /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/dl/ustream-ssl-2024.07.28~99bd3d2b.tar.zst | tar -C /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/ustream-ssl-mbedtls/ustream-ssl-2024.07.28~99bd3d2b/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/ustream-ssl-mbedtls/ustream-ssl-2024.07.28~99bd3d2b
touch /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/ustream-ssl-mbedtls/ustream-ssl-2024.07.28~99bd3d2b/.prepared_a666c3eebf6946f85b105b4493acfc72_18f1e190c5d53547fed41a3eaa76e9e9
rm -f /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/ustream-ssl-mbedtls/ustream-ssl-2024.07.28~99bd3d2b/.configured_*
rm -f /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.ustream-ssl.mbedtls_installed
mkdir -p /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/ustream-ssl-mbedtls/ustream-ssl-2024.07.28~99bd3d2b
(cd /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/ustream-ssl-mbedtls/ustream-ssl-2024.07.28~99bd3d2b; CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/ustream-ssl-mbedtls/ustream-ssl-2024.07.28~99bd3d2b=ustream-ssl-2024.07.28~99bd3d2b -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/ustream-ssl-mbedtls/ustream-ssl-2024.07.28~99bd3d2b=ustream-ssl-2024.07.28~99bd3d2b -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " LDFLAGS="-L/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/lib -L/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro " cmake --no-warn-unused-cli -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=aarch64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/bin/aarch64-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/lib -L/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/lib -L/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/lib -L/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/usr;/home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INST [... cut ...]
-- Generating done (0.0s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
make[3]: *** [Makefile:75: /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/ustream-ssl-mbedtls/ustream-ssl-2024.07.28~99bd3d2b/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
`

## Context (Environment)
I was trying to build for BPI-R4 and then look into the BE14 related patches.

## Specifications
  - native Ubuntu 22.04.3 LTS build env.
  - OpenMPTCProuter platform: BPI-R4

tiborkiss, 2025-08-21:
Just now I had time to post a possible workaround.

I edited .kconfig to have these values:
`
CONFIG_PACKAGE_libmbedtls=y
CONFIG_PACKAGE_libustream-mbedtls=y
# CONFIG_PACKAGE_libustream-openssl is not set
# CONFIG_PACKAGE_libustream-wolfssl is not set
`

then 

`
make package/libs/mbedtls/clean
make package/libs/ustream-ssl/clean

make defconfig

make package/libs/mbedtls/compile V=s
make package/libs/ustream-ssl/compile V=s
`

then finally:

`
make -j1 V=s
`

All the above done in `bpi-r4/6.6/source/` 

The build was successful, but I cannot test it until tonight.

In my opinion, these defaults has to come by default, to not get build failures.

tiborkiss, 2025-08-21:
> It's due to a problem in package dependency. I will commit fix soon.

Thank you for the quick reply.

(Initially I tried the omr_feed_src master, that already failed due to inexistent rust build in CI distribution. They pulled off that older packages.)
Anyway, for my test I am happy that finally I could build. Hopefully I will do some investigations related to BPI-R4-NIC-BE14 patches. Now I have two BPI-R4 with BE14, it's easier to experiment with the second one.

If you committed, you can just close this issue, then I also see if it is ready the fix.

tiborkiss, 2025-08-22:
Finally I could make some tests.
I booted the newly build image and I could not find uhttpd and many other packages.
dmesg and logread has no error.
but when I tried `opkg list-installed` I saw errors in the last 5 lines.

The issue is that the /etc/opkg/distfeeds.conf is broken. Here is what I have in it:
```
src/gz openmptcprouter_core http://<!DOCTYPE html><html lang=/targets/mediatek/filogic/packages
src/gz openmptcprouter_base http://<!DOCTYPE html><html lang=/packages/aarch64_cortex-a53/base
src/gz openmptcprouter_luci http://<!DOCTYPE html><html lang=/packages/aarch64_cortex-a53/luci
src/gz openmptcprouter_openmptcprouter http://<!DOCTYPE html><html lang=/packages/aarch64_cortex-a53/openmptcprouter
src/gz openmptcprouter_packages http://<!DOCTYPE html><html lang=/packages/aarch64_cortex-a53/packages
```

Looking into the build folder, I see:
```
./bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/opkg/distfeeds.conf
./bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/root.orig-mediatek/etc/opkg/distfeeds.conf
./bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/base-files/.pkgdir/base-files/etc/opkg/distfeeds.conf
./bpi-r4/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/opkg/distfeeds.conf
```
and all are the same wrong.

From the `build.log` I extracted the relevant command which makes this:

```
mkdir -p /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/base-files/ipkg-aarch64_cortex-a53/base-files/etc/opkg; ( echo 'src/gz %d_core %U/targets/%S/packages'; echo 'src/gz %d_base %U/packages/%A/base'; echo 'src/gz %d_luci %U/packages/%A/luci'; echo 'src/gz %d_openmptcprouter %U/packages/%A/openmptcprouter'; echo 'src/gz %d_packages %U/packages/%A/packages'; ) >> /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/base-files/ipkg-aarch64_cortex-a53/base-files/etc/opkg/distfeeds.conf; /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/staging_dir/host/bin/sed -i -e 's,%U,http://<!DOCTYPE html><html lang=,g' -e 's,%V,v0.63-snapshot-6.6,g' -e 's,%v,\Lv0.63-snapshot-6.6,g' -e 's,%C,r0+28627-0b392b925f,g' -e 's,%c,\Lr0+28627-0b392b925f,g' -e 's,%D,openmptcprouter,g' -e 's,%d,\Lopenmptcprouter,g' -e 's,%R,r0+28627-0b392b925f,g' -e 's,%T,mediatek,g' -e 's,%S,mediatek/filogic,g' -e 's,%A,aarch64_cortex-a53,g' -e 's,%t,busybox override,g' -e 's,%M,OpenWrt,g' -e 's,%m,https://openwrt.org/,g' -e 's,%b,https://bugs.openwrt.org/,g' -e 's,%u,https://openwrt.org/,g' -e 's,%s,https://forum.openwrt.org/,g' -e 's,%P,Generic,g' -e 's,%h,v0,g' -e 's,%B,1745693631,g' /home/tibor/w/openmptcprouter/bpi-r4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/base-files/ipkg-aarch64_cortex-a53/base-files/etc/opkg/distfeeds.conf
```

Ysurac, 2025-08-22:
You don't have dependencies due to the dependency problem. Sync again the feed tree and make sure "openmptcprouter-full" is enabled via "make menuconfig" in source folder.
All snapshot and images are build with Github CI as you can see here: https://github.com/Ysurac/openmptcprouter-feeds/actions with this basic config: https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/.circleci/config.yml