Ysurac/openmptcprouter #4050: VPS Script is failing on clean Debian12

iEddie-cmd, 2025-08-22:
onnecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/Ysurac/glorytun/tar.gz/refs/heads/tcp [following]
--2025-08-22 12:10:35--  https://codeload.github.com/Ysurac/glorytun/tar.gz/refs/heads/tcp
Resolving codeload.github.com (codeload.github.com)... 140.82.121.9
Connecting to codeload.github.com (codeload.github.com)|140.82.121.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: '/tmp/glorytun-0.0.35.tar.gz'

/tmp/glorytun-0.0.35.tar.gz                                                        [ <=>                                                                                                                                                                                               ]  18.96K  --.-KB/s    in 0.01s   

2025-08-22 12:10:35 (1.79 MB/s) - '/tmp/glorytun-0.0.35.tar.gz' saved [19418]

aclocal: installing 'm4/pkg.m4' from '/usr/share/aclocal/pkg.m4'
configure.ac:12: installing 'm4/compile'
configure.ac:21: installing 'm4/config.guess'
configure.ac:21: installing 'm4/config.sub'
configure.ac:14: installing 'm4/install-sh'
configure.ac:14: installing 'm4/missing'
Makefile.am: error: required file './AUTHORS' not found
Makefile.am: error: required file './ChangeLog' not found
Makefile.am: installing './INSTALL'
Makefile.am: error: required file './NEWS' not found
Makefile.am: installing './COPYING' using GNU General Public License v3 file
Makefile.am:     Consider adding the COPYING file to the version control system
Makefile.am:     for your code, to avoid questions about which license your project uses
Makefile.am:7: warning: source file 'src/common.c' is in a subdirectory,
Makefile.am:7: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least one source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However, this
automake: behavior may change in a future Automake major version, with object
automake: files being placed in the same subdirectory as the corresponding sources.
automake: You are advised to start using 'subdir-objects' option throughout your
automake: project, to avoid future incompatibilities.
Makefile.am:7: warning: source file 'src/main.c' is in a subdirectory,
Makefile.am:7: but option 'subdir-objects' is disabled
Makefile.am:7: warning: source file 'src/option.c' is in a subdirectory,
Makefile.am:7: but option 'subdir-objects' is disabled
Makefile.am:7: warning: source file 'src/tun.c' is in a subdirectory,
Makefile.am:7: but option 'subdir-objects' is disabled
Makefile.am:7: warning: source file 'src/db.c' is in a subdirectory,
Makefile.am:7: but option 'subdir-objects' is disabled
Makefile.am:7: warning: source file 'src/state.c' is in a subdirectory,
Makefile.am:7: but option 'subdir-objects' is disabled
Makefile.am: installing 'm4/depcomp'
autoreconf: error: automake failed with exit status: 1
root@my-vps:~#

Ysurac, 2025-08-22:
It's from snapshot script. It's normal: it's only a snapshot.

iEddie-cmd, 2025-08-22:
I dont understand. Script is not finishing. HOw can I install the vps then?

closed 2025-11-25 by github-actions[bot]