Ysurac/openmptcprouter #4052: OMR server installed with KERNEL=6.12 has no MPTCP BPF (bpf_red) — how to enable it?

angol71, 2025-08-26:
## Expected Behavior
After installing the OMR server with KERNEL="6.12" on Debian 12, the kernel should support MPTCP BPF schedulers so that:
CONFIG_MPTCP_BPF=y is present in /boot/config-$(uname -r).
/usr/lib/bpf/mptcp_bpf_red.bpf.o (or equivalent) exists.
bpftool struct_ops list shows mptcp_sched_ops … red.
sysctl -w net.mptcp.scheduler=bpf_red succeeds and available_schedulers includes bpf_red.

## Current Behavior
After installing via: 
wget -O - https://www.openmptcprouter.com/server-test/debian12-x86_64.sh | KERNEL="6.12" sh
reboot
the server boots 6.12.41-x64v3-xanmod1, but MPTCP BPF is not available:
sysctl -a | grep '^net\.mptcp\.available_schedulers'
# net.mptcp.available_schedulers = default

sysctl -w net.mptcp.scheduler=bpf_red
# sysctl: setting key "net.mptcp.scheduler": No such file or directory

bpftool struct_ops list | grep -i mptcp
# (no output)

ls -1 /usr/lib/bpf | grep -i mptcp
# ls: cannot access '/usr/lib/bpf': No such file or directory

grep -H 'CONFIG_MPTCP_BPF' /boot/config-$(uname -r)
# (no match)

Checked Status → System log: no errors related to BPF/struct_ops/MPTCP scheduler registration.
Note: on my OMR client (v0.62, kernel 6.6.73 x86_64), bpftool struct_ops list shows MPTCP schedulers including red, and net.mptcp.scheduler=bpf_red works — so redundancy is available there.

Question:
Is the server-test installer with KERNEL="6.12" expected to deploy a server kernel that includes CONFIG_MPTCP_BPF and ships the mptcp_bpf_red.bpf.o object? If yes, which exact server snapshot (path/filename) should be used on Debian 12 so that bpf_red is available on the VPS? If not, what is the recommended way to enable bpf_red on the server side?

## Specifications

  - OpenMPTCProuter version: v0.62, kernel 6.6.73 x86_64
  - OpenMPTCProuter VPS version: 6.12.41-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: hetzner.com
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-08-26:
Yes there is no BPF support on VPS side. This would need to make and maintain my own kernel. I will check how I can automatize that...

angol71, 2025-08-29:
Can I install a snapshot with kernel 6.6.73 on a VPS hosted by Hetzner?

Ysurac, 2025-08-30:
For VPS script ? yes, but why ?

closed 2026-01-05 by github-actions[bot]