Ysurac/openmptcprouter #4054: Latency issue with 2 sub 50ms ISP-s. Ping varies from 100 - 1000ms

ovariakos, 2025-08-26:
## Expected Behavior

Latency issue with 2 sub 50ms ISP-s. Ping varies from 100 - 1000ms

The ping should max out around 100-140ms, this is what I expected with the overhead.
Im experiencing 100-1000ms when pinging constantly. (even with tcping)

OMR v0.62-6.6 r0+28431-92e020b50f

Ysurac, 2025-08-26:
Please complete template.
Try another VPN in System->OpenMPTCProuter, Wizard tab, Advanced settings checkbox.
ICMP Ping use the VPN, but TCP use the proxy that have less latency.
To check latency between connections and VPS, look at status page.

ovariakos, 2025-08-26:
[image]

[image]

ovariakos, 2025-08-26:
The issue seems to persist even if I only leave 1 WAN plugged in.

closed 2025-11-29 by github-actions[bot]