Ysurac/openmptcprouter #4067: WAN2 traffic bypasses Xray (VLESS + Reality) tunnel

3xtr3m3d, 2025-09-10:
## Expected Behavior
All WAN interfaces (WAN1 and WAN2) should route traffic through the Xray (VLESS + Reality) tunnel to the VPS, without leaking directly to the internet.

## Current Behavior
With 2 WAN connections configured:
- WAN1 traffic is routed correctly through the Xray tunnel to the VPS.
- WAN2 traffic bypasses the tunnel and goes directly through the WAN2 interface. ( wan2 public ip never seen on the xray logs in the server )

## Possible Solution
Ensure multipath / policy routing rules apply to both WAN interfaces when using Xray (VLESS + Reality) as the tunnel.
Maybe related to missing `ip rule` for WAN2 or tunnel not binding on both interfaces.

## Steps to Reproduce the Problem
1. Configure OpenMPTCProuter with 2 WAN interfaces.  
2. Enable Xray (VLESS + Reality) tunnel to VPS.  
3. Run traffic tests.  
   - WAN1 → goes through VPS tunnel.  
   - WAN2 → leaks directly via WAN2.  

## Context (Environment)
I am trying to use multipath with 2 WAN connections so that all traffic from both WANs is tunneled through Xray to the VPS. Current setup results in WAN2 bypassing the tunnel.

[image]
[image]
[image]

> root@OpenMPTCProuter:~# ip mptcp monitor
[         CREATED] token=3dcd6bc5 remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=52979 dport=443
[         CREATED] token=dd703c9d remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=52995 dport=443
[         CREATED] token=8f7f088f remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53003 dport=443
[         CREATED] token=e960ffad remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53001 dport=443
[     ESTABLISHED] token=3dcd6bc5 remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=52979 dport=443
[       ANNOUNCED] token=3dcd6bc5 remid=2 daddr6=fe80::4001:aff:fe94:a dport=443
[     ESTABLISHED] token=dd703c9d remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=52995 dport=443
[       ANNOUNCED] token=dd703c9d remid=2 daddr6=fe80::4001:aff:fe94:a dport=443
[  SF_ESTABLISHED] token=3dcd6bc5 remid=0 locid=2 saddr4=192.168.1.202 daddr4=34.***.***.245 sport=33650 dport=443 backup=0 ifindex=4
[     ESTABLISHED] token=8f7f088f remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53003 dport=443
[       ANNOUNCED] token=8f7f088f remid=2 daddr6=fe80::4001:aff:fe94:a dport=443
[     ESTABLISHED] token=e960ffad remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53001 dport=443
[       ANNOUNCED] token=e960ffad remid=2 daddr6=fe80::4001:aff:fe94:a dport=443
[  SF_ESTABLISHED] token=dd703c9d remid=0 locid=2 saddr4=192.168.1.202 daddr4=34.***.***.245 sport=14778 dport=443 backup=0 ifindex=4
[  SF_ESTABLISHED] token=8f7f088f remid=0 locid=2 saddr4=192.168.1.202 daddr4=34.***.***.245 sport=46364 dport=443 backup=0 ifindex=4
[  SF_ESTABLISHED] token=e960ffad remid=0 locid=2 saddr4=192.168.1.202 daddr4=34.***.***.245 sport=20412 dport=443 backup=0 ifindex=4
[         CREATED] token=03dfa194 remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53011 dport=443
[         CREATED] token=08e421c7 remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53027 dport=443
[     ESTABLISHED] token=03dfa194 remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53011 dport=443
[       ANNOUNCED] token=03dfa194 remid=2 daddr6=fe80::4001:aff:fe94:a dport=443
[     ESTABLISHED] token=08e421c7 remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53027 dport=443
[       ANNOUNCED] token=08e421c7 remid=2 daddr6=fe80::4001:aff:fe94:a dport=443
[  SF_ESTABLISHED] token=08e421c7 remid=0 locid=2 saddr4=192.168.1.202 daddr4=34.***.***.245 sport=49938 dport=443 backup=0 ifindex=4
[  SF_ESTABLISHED] token=03dfa194 remid=0 locid=2 saddr4=192.168.1.202 daddr4=34.***.***.245 sport=13384 dport=443 backup=0 ifindex=4
[          CLOSED] token=dd703c9d
[         CREATED] token=e4e0c35a remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53035 dport=443
[     ESTABLISHED] token=e4e0c35a remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53035 dport=443
[       ANNOUNCED] token=e4e0c35a remid=2 daddr6=fe80::4001:aff:fe94:a dport=443
[  SF_ESTABLISHED] token=e4e0c35a remid=0 locid=2 saddr4=192.168.1.202 daddr4=34.***.***.245 sport=54266 dport=443 backup=0 ifindex=4
[          CLOSED] token=03dfa194
[         CREATED] token=db8587f0 remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53047 dport=443
[         CREATED] token=b60582c9 remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53059 dport=443
[          CLOSED] token=e098a5de
[     ESTABLISHED] token=db8587f0 remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53047 dport=443
[       ANNOUNCED] token=db8587f0 remid=2 daddr6=fe80::4001:aff:fe94:a dport=443
[     ESTABLISHED] token=b60582c9 remid=0 locid=0 saddr4=192.168.1.201 daddr4=34.***.***.245 sport=53059 dport=443
[       ANNOUNCED] token=b60582c9 remid=2 daddr6=fe80::4001:aff:fe94:a dport=443
[          CLOSED] token=3e651afa
[          CLOSED] token=4528ae8d
[          CLOSED] token=5c4e2fba
[  SF_ESTABLISHED] token=db8587f0 remid=0 locid=2 saddr4=192.168.1.202 daddr4=34.***.***.245 sport=15196 dport=443 backup=0 ifindex=4
[  SF_ESTABLISHED] token=b60582c9 remid=0 locid=2 saddr4=192.168.1.202 daddr4=34.***.***.245 sport=63410 dport=443 backup=0 ifindex=4

Ysurac, 2025-09-10:
I'm not sure to understand what you are testing and where you can see a leak.
In your configuration you should have both wan in a different subnet and lan in another one to isolate everything.
One interface should also be set as master.

Ysurac, 2025-09-10:
You can try another proxy to check.
What you mean with generated kernel traffic from wan2 ? Yes there is some traffic outside the proxy on both wan to check if link is working, to use the VPS API,... Everything between router and VPS is direct traffic; there is a direct route.