Ysurac/openmptcprouter #4078: OMR-Bypass ignored for UDP traffic when enabling V2Ray/XRay UDP

andrewmackrodt, 2025-09-19:
## Expected Behavior
OMR-Bypass is respected for UDP traffic when using V2Ray/Xray proxy and Enable V2Ray/XRay UDP.

## Current Behavior
UDP traffic ignores OMR-Bypass and is transmitted using the VPN. This was previously reported in https://github.com/Ysurac/openmptcprouter/issues/3535 and is an issue since at least v0.61.

## Steps to Reproduce the Problem
Note: the instructions below are for testing using HTTP/3 (QUIC) but the same behaviour has been tested by running `tcpdump` on a non-OMR VPS and sending a UDP packet using `nc -v -u -z -w 3 <ip> <port>` from a client where the MAC address is in the OMR-Bypass list.

  1. Disable **Firewall - Traffic Rules** for **Block QUIC Proxy** and **Block QUIC All**
  2. Set proxy type to **XRAY VLESS**
  3. Enable **V2Ray/XRay UDP** under advanced settings
  4. Add an IP/Mac OMR-Bypass rule for the test computer, set it to use WAN1, save and apply
  5. Wait a few seconds for the rule to be applied, e.g. `curl ifconfig.co` should show WAN IP
  6. Once new IP is confirmed, use an HTTP/3 supporting browser (e.g. Chrome or Edge) and visit https://www.whatismyip.com/ - you may need to refresh after the first visit. The second and subsequent requests should be made using HTTP/3 (QUIC/UDP) and you will see the OMR VPN IP address. If this cannot be reproduced, refer to step 1 to ensure the router is not blocking QUIC, and open the network tab under developer tools, ensure protocol column is visible and check that the request is made with h3.

## Context (Environment)
I find UDP using XRay VLESS to work more reliably versus using the VPN especially when a client computer connects to a UDP based OpenVPN server or Wireguard.

I have various OMR-Bypass rules for things like Netflix or selective device rules, e.g. before I stream from Xbox Cloud. In the latter case, UDP is essential and disabling QUIC as a traffic rule would not work as a workaround.

## Specifications
  - OpenMPTCProuter version: v0.63-6.12 r0+30806-070d8eb4d5
  - OpenMPTCProuter VPS version: v0.63-6.12
  - OpenMPTCProuter VPS provider: -
  - OpenMPTCProuter platform: RPI5
  - Country: UK

Ysurac, 2025-09-19:
A part if missing for UDP, I will fix that.

Ysurac, 2025-12-31:
@tiborkiss you can test latest snapshot: https://snapshots.openmptcprouter.com/6.12/