Ysurac/openmptcprouter #4086: AES-256-GCM - VPN tunnel DOWN

h45h3ncryp73d, 2025-09-27:
## Expected Behavior
VPN tunnel should work just normal with AES-256 encryption selected (I guess since i3 should have AES integrated)

## Current Behavior
Maybe I'm missing something since I've just upgraded OMR from RPi4b to i3 NUC but I've only imported settings from RPi as I don't believe there is anything why this shouldn't be working.

Anyway when I change encryption from cacha20 to AES-256-GCM VPN tunnel goes down.
cacha20 -> VPN tunnel OK
AES-256-GCM ->VPN tunnel DOWN

If I restart OMR the whole package stops working altogether meaning OMR is completely down and VPS is  constantly switching between direct output and VPS ...

## Specifications

  - OpenMPTCProuter version: v0.63-6.12
  - OpenMPTCProuter VPS version: 0.1039 6.12.41-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: Alwyzon
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-09-27:
What is the VPN you are using ? Glorytun TCP is not working in current release, seems to work in snapshot.

h45h3ncryp73d, 2025-09-27:
Hmmm ... It turned to OK now ...

h45h3ncryp73d, 2025-09-28:
> What is the VPN you are using ? Glorytun TCP is not working in current release, seems to work in snapshot.

Seems like it's not working ... Changing to MLVPN instantly solved the problem.

closed 2026-01-03 by github-actions[bot]